larsimmisch/pyalsaaudio #149: mix and play several wave files at once?

Choons, 2024-05-16:
Hi, thank you for making pyalsaaudio!  Just getting started with it and wondering if pyalsaaudio has the capability to mix and play several audio files at once like in a chord, for instance, or say if used in a game where several sound effects might occur at the same time or overlap?

ossilator, 2024-05-17:
you can try just opening the same device multiple times. usually, that should work.

ideally, the hardware (and the driver) supports this natively, up to some limit.

the next layer is the alsa library. its plugin architecture supports software mixing, in particular via dmix.

lastly, every modern system is running a sound server like pipewire, which does software mixing. usually, the sound server should be exposed as the default alsa device, so things should just work.

so in summary, if it doesn't work, then you system is not configured properly for the use case. pyalsaaudio doesn't have anything to do with that as such, so i'm closing the issue.

closed 2024-05-17 by ossilator

Choons, 2024-05-17:
Update to this: I discovered I can specify 'dmix' instead of 'default' with:
audio_device = alsaaudio.PCM(channels=channels, rate=rate, format=format, periodsize=periodsize, device='dmix')

and pyalsaaudio will play the sound file, but at the wrong speed/pitch as apparently the rate setting has no effect. I can also open additional instances of the device like that. Seems like I'm really close to being able to mix multiple sounds, if I could just get the sample rate set right.

ossilator, 2024-05-18:
you probably need to chain the dmix plugin with the hwplug one, which does the resampling. don't ask me for details, but you'll find plenty of config stuff to learn from in /usr/share/alsa. note that if you can control the sample rate of the source, that would be better than resampling.

Choons, 2024-05-18:
OK, thank you for the help. I think I am going to try handling the mixing on my own with numpy and samples stored in RAM and just send that single pre-mixed stream into the default device as I already know it works properly when instantiated from pyalsaaudio.