larsimmisch/pyalsaaudio #153: Possibility of recording from speakers

s0d3s, 2024-06-16:
Hi, thanks for your repo!
Is it possible to record audio from speakers (select them as input)? If not, is this in the plans?

ossilator, 2024-06-17:
if your device provides some form of loopback function, then sure. play with the mixer settings.

there is nothing to do for us here, so i'm closing the issue.

closed 2024-06-17 by ossilator

RonaldAJ, 2024-06-17:
I agree with closing the issue.

As a rule you cannot use speakers for recording audio. The speakers will be connected to the digital-to-analog converter, whereas a microphone is connected to an analog-to-digital converter (ADC). Nothing stops you from connecting a speaker to an ADC, but I expect it requires special hardware to get it working, performing analog low-pass filtering and amplification of the signal into the range of the ADC used.  It is probably cheaper and faster to buy an extra microphone.