larsimmisch/pyalsaaudio #155: alsaaudio.PCM() not playing nicely with pipewire

prayner, 2024-10-28:
Unsure whether this is a pyalsaaudio or pipewire thing but guidance
greatly appreciated.
I'm running ubuntu 24.04 with pipewire installed.
Here's the buffer from the emacs pipewire command
Devices:
  46: sof-hda-dsp (Audio/Device): Play HiFi quality Music
    47: HDMI / DisplayPort 3 Output (Audio/Sink) 100
      33: playback_FL 100
      40: playback_FR 100
      41: monitor_FR 100
      42: monitor_FL 100
    48: HDMI / DisplayPort 2 Output (Audio/Sink) 100
      39: playback_FL 100
      53: monitor_FL 100
      54: playback_FR 100
      55: monitor_FR 100
    49: HDMI / DisplayPort 1 Output (Audio/Sink) 100
      56: playback_FL 100
      57: monitor_FL 100
      58: playback_FR 100
      59: monitor_FR 100
    50: Speaker + Headphones (Audio/Sink) 100
      60: playback_FL 59
      61: monitor_FL 100
      62: playback_FR 59
      63: monitor_FR 100
    51: Headphones Stereo Microphone (Audio/Source) 100
      64: capture_FL 4
      65: capture_FR 4
    52: Digital Microphone (Audio/Source) 100
      66: capture_FL 100
      67: capture_FR 100
Clients:
  32: pipewire
  34: WirePlumber
  35: WirePlumber [export]
  68: GNOME Volume Control Media Keys
  69: speech-dispatcher-dummy
  70: GNOME Shell Volume Control
  71: gnome-shell
  74: gsd-power
  77: speech-dispatcher-voxin
  79: Google Chrome input
  83: xdg-desktop-portal
  84: emacs
 125: Mutter
 127: PipeWire ALSA [tclsh8.6]
 330: pw-cli
 367: pactl
this all works fine. the tclsh client is the speech server from the
emacspeak package which is the main input.

Running ipython:
ipython
In [1]: import alsaaudio

In [2]: alsaaudio.pcms()
Out[2]: 
['null',
 'pipewire',
 'default',
 'hw:CARD=sofhdadsp,DEV=0',
 'hw:CARD=sofhdadsp,DEV=3',
 'hw:CARD=sofhdadsp,DEV=4',
 'hw:CARD=sofhdadsp,DEV=5',
 'hw:CARD=sofhdadsp,DEV=31',
 'plughw:CARD=sofhdadsp,DEV=0',
 'plughw:CARD=sofhdadsp,DEV=3',
 'plughw:CARD=sofhdadsp,DEV=4',
 'plughw:CARD=sofhdadsp,DEV=5',
 'plughw:CARD=sofhdadsp,DEV=31',
 'sysdefault:CARD=sofhdadsp',
 'dmix:CARD=sofhdadsp,DEV=0',
 'dmix:CARD=sofhdadsp,DEV=3',
 'dmix:CARD=sofhdadsp,DEV=4',
 'dmix:CARD=sofhdadsp,DEV=5',
 'dmix:CARD=sofhdadsp,DEV=31']
and here's the output from aplay -lL
aplay -lL
null
    Discard all samples (playback) or generate zero samples (capture)
pipewire
    PipeWire Sound Server
default
    Default ALSA Output (currently PipeWire Media Server)
hw:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=3
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=4
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=5
    sof-hda-dsp, 
    Direct hardware device without any conversions
hw:CARD=sofhdadsp,DEV=31
    sof-hda-dsp, 
    Direct hardware device without any conversions
plughw:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=3
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=4
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=5
    sof-hda-dsp, 
    Hardware device with all software conversions
plughw:CARD=sofhdadsp,DEV=31
    sof-hda-dsp, 
    Hardware device with all software conversions
sysdefault:CARD=sofhdadsp
    sof-hda-dsp, 
    Default Audio Device
dmix:CARD=sofhdadsp,DEV=0
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=3
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=4
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=5
    sof-hda-dsp, 
    Direct sample mixing device
dmix:CARD=sofhdadsp,DEV=31
    sof-hda-dsp, 
    Direct sample mixing device
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 31: Deepbuffer HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

So as I understand it calling
out=alsaaudio.PCM(device='default')
should open a connection to the pipewire media server and handle all
the resampling etc.
This isn't working, it produces endless sound loops from the tclsh
server above (destroying speech output) and won't go on to produce
sound from the rest of the application. This has been working with
pulseaudio for a number of years. Are there things I need to do to get
pyalsaaudio and pipewire talking?

ossilator, 2024-10-29:
> Are there things I need to do to get pyalsaaudio and pipewire talking?

pyaa is really a "rather boring" alsa client, so nothing special should be required.

try taking pyaa out of the loop: try actually using aplay and arecord, with different parameters. does it work?

export LIBASOUND_DEBUG=1 and compare the output between working and not working constellations.

prayner, 2024-11-12:
I can demonstrate the problem on my system with the supplied isine.py.
Here's a modified version. Running this on my system produces a pure
tone as it should but with lots of breaks and static

''' Use this example from an interactive python session, for example:

>>> from isine import change
>>> change(880)
'''

from __future__ import print_function

import sys
import time
from threading import Thread
from multiprocessing import Queue

if sys.version_info[0] < 3:
    from Queue import Empty
else:
    from queue import Empty

from math import pi, sin, ceil
import struct
import itertools
import alsaaudio

sampling_rate = 48000

format = alsaaudio.PCM_FORMAT_S16_LE
pack_format = 'h'  # short int, matching S16
channels = 2

def nearest_frequency(frequency):
    # calculate the nearest frequency where the wave form fits into the buffer
    # in other words, select f so that sampling_rate/f is an integer
    return float(sampling_rate)/int(sampling_rate/frequency)

def generate(frequency, duration = 0.125):
    # generate a buffer with a sine wave of `frequency`
    # that is approximately `duration` seconds long

    # the length of a full sine wave at the frequency
    cycle_size = int(sampling_rate / frequency)

    # number of full cycles we can fit into the duration
    factor = int(ceil(duration * frequency))

    # total number of frames
    size = cycle_size * factor

    sine = [ int(12767 * sin(2 * pi * frequency * i / sampling_rate)) \
             for i in range(size)]

    if channels > 1:
        sine = list(itertools.chain.from_iterable(itertools.repeat(x, channels) for x in sine))

    return struct.pack(str(size * channels) + pack_format, *sine)

class SinePlayer(Thread):

    def __init__(self, frequency = 440.0):
        Thread.__init__(self, daemon=True)
        self.device = alsaaudio.PCM(channels=channels, format=format, rate=sampling_rate)
        self.queue = Queue()
        self.change(frequency)

    def change(self, frequency):
        '''This is called outside of the player thread'''
        # we generate the buffer in the calling thread for less
        # latency when switching frequencies

        if frequency > sampling_rate / 2:
            raise ValueError('maximum frequency is %d' % (sampling_rate / 2))

        f = nearest_frequency(frequency)
        print('nearest frequency: %f' % f)

        buf = generate(f)
        self.queue.put(buf)

    def run(self):
        buffer = None
        while True:
            try:
                buffer = self.queue.get(False)
            except Empty:
                pass
            if buffer:
                if self.device.write(buffer) < 0:
                    print("Playback buffer underrun! Continuing nonetheless ...")

isine = SinePlayer()
isine.start()

time.sleep(1)
isine.change(1000)
time.sleep(5)
sys.exit()

ossilator, 2024-11-12:
so you reduced the amplitude and increased the play time.
that helps nothing to diagnose the problem.
please follow the instructions i gave previously.

ossilator, 2025-02-25:
no info forthcoming.

closed 2025-02-25 by ossilator